luc-github/ESP3D-WEBUI #282: [FEATURE REQUEST]I want to implement GRBL's online teaching function!

wocut, 2022-12-19:
I want to implement GRBL's online teaching function! I use the jog function to move the tool to start 1 and record this point; Then move the tool to end 2 and record this point; The program wrote my whole process into G-code!

luc-github, 2022-12-19:
so what is your question ? How do you want to implement it?

wocut, 2022-12-19:
Thanks for your reply! 

I wanted to edit the index .html, I opened the index .html with vscode,  I used show preview, but I couldn't see anything in the Microsoft Edge! 
What software should I use to edit this index.html file?

luc-github, 2022-12-19:
It does not work like this, depending of which version of webui , sources and build process are diferent.
* for 2.1 sources are in https://github.com/luc-github/ESP3D-WEBUI/tree/2.1/www and build process is https://github.com/luc-github/ESP3D-WEBUI/wiki/Compilation---Development

* For 3.0 sources are in https://github.com/luc-github/ESP3D-WEBUI/tree/3.0/src and build process is https://github.com/luc-github/ESP3D-WEBUI/tree/3.0/src

index.html is minified and gzip to be as small as possible you cannot edit it easily

wocut, 2022-12-20:
[image]
Hello!
I opened your link: https://github.com/luc-github/ESP3D-WEBUI/tree/3.0#build-indexhtmlgz-to-dist-folder
I opened this folder with vscode, I clicked Show Preview, but I only see a blank sheet in the Microsoft Live Preview window!
[image]

What software do I use to edit this index .html?
Is it OK to use vscode?

luc-github, 2022-12-25:
@wocut is your question answered? can we close issue ?

closed 2022-12-31 by luc-github