luc-github/ESP3D-WEBUI #290: [BUG] Saving fails on a clean flash where there is no pre-existing preferences.json

errolt, 2023-01-05:
**Describe the bug**
Saving settings fails with: "Error saving data"

**To Reproduce**
Board: MKS DLC32.
Do a complete flash erase: idf.py -p /dev/ttyUSB0 erase_flash
Flash grblHAL ESP32: idf.py -p /dev/ttyUSB0 flash monitor
Connect grlbHAL to wifi via $74 and $75 commands.
Upload ESP32-WebUI index.html.gz to flash
Now, if you chance anything on the Setting page and try to save you get the error message.
If you export the json, rename it to preferances.json and upload it to the flash FS, then saving works.
PS. Also gets this message on connection: 404 - File not found preferences.json

**Expected behavior**
Able to save settings on a new install where there is no existing preferences.json

**Screenshots**
[image]

**ESP3D Firmware:  (please complete the following information)**
- grblHAL version 1.1f
- Web UI Version: 3.0
- Wifi mode: STA
- Any change done in FW: None

**Target Firmware:  (please complete the following information)**
 - Name: grblHAL ESP32?
 - Version 1.1f?

**Board used (please complete the following information):**
 - MCU: ESP32
 - Name: MKS DLC32
 - Flash size: 8M

**Browser used (please complete the following information):**
- Name: Firefox
- Version: 106  
- Operating system: Ubuntu 20.04

**Additional context**
Start with a clean flash.

luc-github, 2023-01-05:
no preferences.json is not an issue , it will be generated as soon as you save the settings

errolt, 2023-01-05:
Unfortunately that doesn't happen. I get "Error saving data", as depicted in the image.

luc-github, 2023-01-05:
Please provide proper version of webui and grblhal - click on grblHAL image they should be displayed

errolt, 2023-01-05:
Ahh. 
[image]

luc-github, 2023-01-05:
I have older version of  grblHAL and I have no issue - I need to update to see if I can reproduce your issue

luc-github, 2023-01-05:
can you also share your CMakeLists.txt you used to configure grblHAL

errolt, 2023-01-05:
[CMakeLists.txt](https://github.com/luc-github/ESP3D-WEBUI/files/10351672/CMakeLists.txt)

Don't know if this would be useful:

```
$i
ok
[VER:1.1f.20221115:]
[OPT:VNMSL,35,1024,3,0]
[AXS:3:XYZ]
[NEWOPT:ENUMS,RT-,NOPROBE,REBOOT,TC,SED,RTC,WIFI,FTP,SD]
[FIRMWARE:grblHAL]
[NVS STORAGE:*FLASH]
[DRIVER:ESP32]
[DRIVER VERSION:221205]
[DRIVER OPTIONS:v4.3.4-309-g389a848a6a]
[BOARD:MKS DLC32 2.0]
[WIFI MAC:4C:75:25:6D:42:1C]
[IP:192.168.1.122]
[NETCON:Websocket]
[PLUGIN:WebUI v0.13]
[PLUGIN:SDCARD v1.07]
ok
```

errolt, 2023-01-05:
Am I doing something dumb?

Some settings save successfully, like "steps/mm", or "step direction invert". 
But others like "soft limits enable" causes a save error.

luc-github, 2023-01-05:
sorry you save which part ? Interface or Feature or GRBL, preferences.json is only for interface 

Can you clarify 
I cannot duplicate on  BOARD_BDRING_V3P5 "Compile for v3.5 3-axis board"

[image]

Do you do different than the screen cast I just did ?

I will try to setup and DLC32 later - I should have one - but it should not be hardware dependent issue

errolt, 2023-01-05:
Never mind. I'm sorry. I didn't see the three tabs on the settings screen.
If I go to the second tab(Interface) and save then it does create a preferences.json file.
I was confused in thinking that software limits would be part of the UI.

Sorry that I wasted your time.

luc-github, 2023-01-05:
but you mentioned  the error saving data - it is doing what ?

errolt, 2023-01-05:
Maybe a feature not enabled in grblHAL?

If I go to settings, Features, and set "Soft Limits Enable" to Enabled, and save then "Error saving data". But this would probably be an issue with grblHAL, and probably not ESP3D-WEBUI.

It was just one of the settings I used to test, as it wouldn't cause hardware problems if I enable it.

luc-github, 2023-01-05:
Ok I reproduce your issue enabling software limit 

@trejeio
I got 
`{"cmd":"401","status":"error","data":"Set failed"}`
When using webui setting or manual command 

`[ESP401]P=20#0 T=B V=1 json=yes`

[ESP400] give consistent
```
 {
   "F": "Limits/Limits",
   "P": "20#0",
   "T": "B",
   "V": "0",
   "H": "Soft limits enable",
   "O": [
    {
     "Enabled": "1"
    },
    {
     "Disabled": "0"
    }
   ]
  },
```
Any idea ?

luc-github, 2023-01-05:
it may be a missing boolean emulation ?

luc-github, 2023-01-05:
@terjeio it may be a missing boolean emulation ?

terjeio, 2023-01-05:
The error is returned from $20 (enable soft limits) when homing is not enabled.
Perhaps I should return the textual representation of the error code instead of the generic "Set failed"?
For this it would be "Soft limits cannot be enabled without homing also enabled."

luc-github, 2023-01-05:
yes it will be displayed in toaster and in notifications list.

luc-github, 2023-01-07:
ok so  I can close issue then

closed 2023-01-07 by luc-github