luc-github/ESP3D-WEBUI #294: [BUG] temperature panel in configuration

atsju, 2023-01-18:
**Describe the bug**
This panel in configuration tab does not correspond to any "temperature panel" in dashboard. Is it normal ?
[image]

**ESP3D Firmware:  (please complete the following information)**
- Version: v3.0.0-alpha3
- Web UI Version:v3.0.0-alpha3
- Wifi mode:client
- Any change done in FW: none

**Target Firmware:  (please complete the following information)**
 - Name:marlin
 - Version 2.1.2

**Board used (please complete the following information):**
 - MCU: esp32cam
 - Name:[e.g. NodeMCU2s] 
 - Flash size: [e.g. 4M: 2M/2M]

**Browser used (please complete the following information):**
- Name: chrome
- Version  [e.g. 2.0]
- Operating system  Windows 10

**Additional context**
Add any other context about the problem here.
Developer tools output, error message, etc...

luc-github, 2023-01-18:
what do you mean?

atsju, 2023-01-19:
OK this was the issue

luc-github, 2023-01-20:
Ok I close issue then

closed 2023-01-20 by luc-github