luc-github/ESP3D-WEBUI #299: [BUG]Error message not displayed in Notifications, followup to #290

terjeio, 2023-01-27:
**Describe the bug**
Error message reported when attempt to save setting is not displayed in Notifications as suggested in [#290](https://github.com/luc-github/ESP3D-WEBUI/issues/290#issuecomment-1372169371):

BTW What is/where can I find the "toaster" you mentioned in [#290](https://github.com/luc-github/ESP3D-WEBUI/issues/290#issuecomment-1372169371)?

**To Reproduce**
Steps to reproduce the behavior:
Trying to enable soft limits without enabling homing.

**Screenshots**

[image]

**ESP3D Firmware:  (please complete the following information)**
- Web UI Version: 3.0.0-a27.GH2

**Target Firmware:  (please complete the following information)**
 - Name: grblHAL
 - Version: 20230126

luc-github, 2023-01-27:
the toaster appear on top of webpage during 5 seconds

luc-github, 2023-01-27:
@terjeio my bad - I missed to copy the data content in error message - should be fixed in https://github.com/luc-github/ESP3D-WEBUI/commit/f5275d4b4fc1a5bcfd287b19bdfa9110dbcb0e41

terjeio, 2023-01-27:
Thanks, working now!

[image]

luc-github, 2023-01-27:
thank you for the confirmation ^_^

luc-github, 2023-02-01:
I close issue then

closed 2023-02-01 by luc-github