luc-github/ESP3D-WEBUI #300: [FEATURE REQUEST] Please add several additional CNC file extension types.

phil-barrett, 2023-02-09:
Some default file types (ie extensions) of CAD/CAM (ie, GCode) files are not listed in the Files panel/Extensions list.  Thus download will not show them by default and, if downloaded, they aren't marked with the Play file icon.

Specifically, the extensions are .ngc - used by F-Engrave and .cnc used by several CAD/CAM tools.

Please add ;cnc;ngc to the end of the text in the Extensions box of the ESP3D Interface panel in Settings.

The alternative is for each user to manually add these extensions to the list and click on refresh. This will happen after they come here, possibly post a bug report that WebUI can't find their files and get told what to do.

luc-github, 2023-02-09:
go to settings of webui and add them, this is configurable

luc-github, 2023-02-09:
what version are you using?

phil-barrett, 2023-02-09:
3.0.0-a30.GH2

Sure, I can add them to mine but I was hoping that it gets in the release version so all the users wouldn't need to do it, too.  The process they will go through is:
- look to upload their gcode file
- not see it.  complain or maybe figure out that they need to use All Files (*.*)
- upload the file
- see that it doesn't have the Play icon next to it.  complain.
- get told to add the extension and refresh.

luc-github, 2023-02-09:
ok understood
only these 2 extensions are missing ? any current extensions is not relevant?

phil-barrett, 2023-02-09:
> only these 2 extensions are missing ? any current extensions is not relevant?

That's a good question. I've used around 10 different CAD/CAM tools so not that many considering how many are out there.  I can not say which are safe to drop.  There is, literally, no standardization out there. And, to make it worse, some CAD/CAM apps allow users to pick file extensions in their POST configuration code.  Yuck.

luc-github, 2023-02-09:
@phil-barrett extensions added - I did not built new packages yet - I have others changes to add soon, I will rebuild them at this moment
I keep FR open until packages are built

luc-github, 2023-02-13:
packages have been built

commit e4cd6d64 (2023-02-09): Fix #300
closed 2023-02-13 by luc-github