luc-github/ESP3D-WEBUI #336: Does not load the printer settings in the merlin/printer tab.

dreamer1976, 2023-02-26:
I use ESP3d V3 (ESP3D 3.0 alpha stage) and have tried different WUI but I keep getting 1 ERROR.OK message when I try to get the settings under settings and then merlin(printer).
after the error I don't see anything in the merlin tab of the printer settings.
If I look in my dashboard at terminal, I see that settings have been retrieved.

***output terminal***
M503
echo:  G21    ; Units in mm (mm)
echo:Filament settings: Disabled
echo:  M200 D1.75
echo:  M200 D0
echo:Steps per unit:
echo: M92 X80.00 Y80.00 Z400.00 E93.00
echo:Maximum feedrates (units/s):
echo:  M203 X500.00 Y500.00 Z5.00 E25.00
echo:Maximum Acceleration (units/s2):
echo:  M201 X500.00 Y500.00 Z100.00 E1000.00
echo:Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
echo:  M204 P500.00 R1000.00 T1000.00
echo:Advanced: B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> X<max_x_jerk> Y<max_y_jerk> Z<max_z_jerk> E<max_e_jerk>
echo:  M205 B20000.00 S0.00 T0.00 X10.00 Y10.00 Z0.30 E5.00
echo:Home offset:
echo:  M206 X0.00 Y0.00 Z0.00
echo:Auto Bed Leveling:
echo:  M420 S0 Z0.00
echo:  G29 W I0 J0 Z0.35250
echo:  G29 W I1 J0 Z0.15350
echo:  G29 W I2 J0 Z0.10850
echo:  G29 W I0 J1 Z0.08750
echo:  G29 W I1 J1 Z-0.00100
echo:  G29 W I2 J1 Z0.02350
echo:  G29 W I0 J2 Z-0.09000
echo:  G29 W I1 J2 Z-0.19850
echo:  G29 W I2 J2 Z-0.16250
echo:PID settings:
echo:  M301 P28.72 I2.62 D78.81
echo:  M304 P462.10 I85.47 D624.59
echo:Power-Loss Recovery:
echo:  M413 S1
echo:Z-Probe Offset (mm):
echo:  M851 X-45.00 Y-20.00 Z-2.98
ok

Can someone tell me what I am missing or doing wrong. I am using an ender 3 V2.
If I control the printer via the UI, the printer responds

***Terminal when i use JOG in the UI and the printer responds***
G91
G1 Y+10 F1000
G90
ok
ok
ok
G91
G1 Y+10 F1000
G90
ok
ok
ok

thank you in advance

luc-github, 2023-02-26:
what printer board are you using?
what esp board are you using?
what Marlin version are you using?

dreamer1976, 2023-02-26:
webui  through this link https://luc-github.github.io/#/config/features
ESP board is a ESP32-CAM
Merlin version i think 1.3.1

I previously had ESP3D on a ESP8622 with ESP3d V2.1 and it just worked fine there and also got the printer settings.
But i want to use a camera.

luc-github, 2023-02-26:
do M115 to know the Marlin version

dreamer1976, 2023-02-26:
FIRMWARE_NAME:Marlin Ver 1.3.1 SOURCE_CODE_URL:https://github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:Ender-3 V2 EXTRUDER_COUNT:1 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff
Cap:SERIAL_XON_XOFF:0
Cap:BINARY_FILE_TRANSFER:0
Cap:EEPROM:1
Cap:VOLUMETRIC:1
Cap:AUTOREPORT_TEMP:1
Cap:PROGRESS:0
Cap:PRINT_JOB:1
Cap:AUTOLEVEL:1
Cap:Z_PROBE:1
Cap:LEVELING_DATA:1
Cap:BUILD_PERCENT:0
Cap:SOFTWARE_POWER:0
Cap:TOGGLE_LIGHTS:0
Cap:CASE_LIGHT_BRIGHTNESS:0
Cap:EMERGENCY_PARSER:0
Cap:PROMPT_SUPPORT:0
Cap:AUTOREPORT_SD_STATUS:0
Cap:THERMAL_PROTECTION:1
Cap:MOTION_MODES:0
Cap:CHAMBER_TEMPERATURE:0
ok

luc-github, 2023-02-26:
Marlin 1.X has different output than 2.X and in ESP3D V3 I only tested Marlin 2.X
```
M503
echo:; Linear Units:
echo:  G21 ; (mm)
echo:; Temperature Units:
echo:  M149 C ; Units in Celsius
echo:; Filament settings (Disabled):
echo:  M200 S0 D1.75
echo:; Steps per unit:
echo:  M92 X80.00 Y80.00 Z400.00 E500.00
echo:; Max feedrates (units/s):
echo:  M203 X300.00 Y300.00 Z5.00 E25.00
echo:; Max Acceleration (units/s2):
echo:  M201 X3000.00 Y3000.00 Z100.00 E10000.00
echo:; Acceleration (units/s2) (P<print-accel> R<retract-accel> T<travel-accel>):
echo:  M204 P3000.00 R3000.00 T3000.00
echo:; Advanced (B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> J<junc_dev>):
echo:  M205 B20000.00 S0.00 T0.00 J0.01
echo:; Home offset:
echo:  M206 X0.00 Y0.00 Z0.00
echo:; Hotend PID:
echo:  M301 P22.20 I1.08 D114.00
ok
```

luc-github, 2023-02-26:
I will check if can handle also 1.X  tomorrow

dreamer1976, 2023-02-26:
Thank you very much so far

luc-github, 2023-02-26:
Please give a try and feedback on this one
[index.html.gz](https://github.com/luc-github/ESP3D-WEBUI/files/10833768/index.html.gz)

dreamer1976, 2023-02-26:
Okay that works, I now get to see the pinter info.
only nothing in the dashboard yet or is that correct?

Did you change anything via the webUI configuration site or manually?

luc-github, 2023-02-26:
you need to go to settings and add polling commands like M105 and M114

dreamer1976, 2023-02-26:
Okay that works too. is there a tutorial/manual that I can use for set up?
Anyway, thank you very much, now I can move on.

luc-github, 2023-02-26:
no there is none yet, I had no time for it

dreamer1976, 2023-02-26:
I understand but I can continue playing. thank you

luc-github, 2023-02-27:
Thank you for the feedback I have updated git repository

closed 2023-02-27 by luc-github