luc-github/ESP3D-WEBUI #337: Individual language files do not seem to work for 2.1

3D-Darryl, 2023-02-26:
Using Esp3D 2.1.2 with ESP32 (from Universal Solder) and it works with the 162K index.html.gz, but if I load English or Italian versions (about 28K) using the fallback page, it fails and falls back to the fallback page.

Possibly related, using the original 162K index.html.gz, If I select the camera page and go to https://192.168.0.1/flashview.html (in Access Point mode) it times out and disconnects.

Is there still an online version of the configurator or do i have to find a linux machine to build the sources?

luc-github, 2023-02-26:
english only is about 108KB not 28KB
https://github.com/luc-github/ESP3D-WEBUI/raw/2.1/languages/en/index.html.gz

italian is about 112KB https://github.com/luc-github/ESP3D-WEBUI/raw/2.1/languages/it/index.html.gz

you did not download the proper index.html.gz

and there is no configurator for version 2.x

3D-Darryl, 2023-02-26:
Thank you for the GZ files.  I am far from a github expert.  Some pages have DOWNLOAD buttons, the language page does not.  I was suspicious of the files.

What is the correct way to access the camera?  I was surprised that I had  to tell it where to look since it is on the same board.  Is 192.168.0.1/flashview.html supposed to work (in AP mode)?

luc-github, 2023-02-27:
esp3d 2.1 only support external ip camera, just put ip in camera page input

luc-github, 2023-02-27:
I close issue then

closed 2023-02-27 by luc-github

3D-Darryl, 2023-02-27:
Sorry to hear that you only support external camera.  My only reason for changing from ESP8266 to ESP32 was the camera.  Any chance of support in 3.0 version?

luc-github, 2023-02-27:
did you read what I wrote ?

3D-Darryl, 2023-02-27:
Sorry, i was responding to  an older email.   I saw a lot of bug reports on the 3.0 Alpha version , I was afraid to try it.

luc-github, 2023-02-27:
well 3.0 was very ambitious but it is running for a while now and of course fixing bugs 
the alpha state is mainly due to missing features not to existing
https://github.com/luc-github/ESP3D/blob/3.0/README.md#todo-list-to-move-to-beta

3D-Darryl, 2023-02-27:
Build of 3.0 fails because it cannot open SdFat.h.  Copied SdFat2 folder from extra-libraries over to Arduino libraries but still fails "no such file or directory".

luc-github, 2023-02-27:
this ssue is now closed - please open proper issue in https://github.com/luc-github/ESP3D/