luc-github/ESP3D-WEBUI #340: Jog Rose size

Ulisse76, 2023-03-06:
is it possible to insert a combobox/radiobutton to increase the size of the JogRose by x1 x2 x4 x8 times?
TY

luc-github, 2023-03-06:
what do you mean?
do you have screenshot?

Ulisse76, 2023-03-07:
I attach the screenshot. compared to the box the "JogRose" has reduced dimensions. it would be nice to have it in a larger (double for example) size for those who use touchscreens

[image]

luc-github, 2023-03-07:
the area you mention is reserved to macro buttons in 2.x and UI is not maid to be dynamic enough to be resized, so no it won't happen sorry
 
but in webui 3.X the UI is the following :
[image]

and it has now a mobile view for phone/tablet
[image]

closed 2023-03-07 by luc-github

Ulisse76, 2023-03-07:
I found a solution that may be useful to someone. I zoomed in 1.4 times like this:

<svg width="310" height="**400**" xmlns="http://www.w3.org/2000/svg" version="1.1" **style="transform: scale(1.4); transform-origin: top left;"**>

in this way I am happy and I solved it on the fly

DivingDuck, 2023-03-07:
Hi there,
@luc-github , wouldn't it be better to increase the svg width so that is fits a bit better in the jog container? The default value is quite small compared to the jog panel width and this is why it looks a little bit lost in a browser compared to other panels like for Macros, Camera or Charts.

svg width=250 (original)
[image]

svg width=340 (modified)
[image]

That way it works as well with the zoom function in a browser and still usable with 50% zoom factor in a browser.

File jog.js, line 502:  width="250px" to width="340px" 
or what ever fit better (maybe something between 300 and 340).

luc-github, 2023-03-07:
indeed this size come from original width panel and I forget to adjust following new panel width, I have added a css to auto adjust  width in case we change panel min width again which is currently 330px and removed the hard coded width in svg