luc-github/ESP3D-WEBUI #341: [BUG] no opening WEBUI after upload index.html.gz file

szczarek, 2023-03-07:
**Describe the bug**
After upload a index.html.gz file on 192.168.0.1?forcefallback=yes adress, i can`t go to a WEBUI side on 192.168.0.1 and do anything with ESP3D. 

**To Reproduce**
Steps to reproduce the behavior:
1. open 192.168.0.1?forcefallback=yes
2. upload index.html.gz file
3. go to 192.168.0.1
4. no reaction, blank screen.

**ESP3D Firmware: 
- Web UI Version:[3.0]

**Target Firmware:  
 - Name: [e.g. Marlin ]
 - Version [e.g. 2.1.1]

**Board used 
 - MCU: [ESP32] 
 - Name:[ESP32CAM] 
 - Flash size: [4M]

**Browser used
- Name: [opera]
- Operating system [Windows 10]

luc-github, 2023-03-07:
can you do a screenshot showing you cannot do anything?
I do not understand

szczarek, 2023-03-08:
It`s side of upload
[image]
and the side after click ESP3D button
[image]

szczarek, 2023-03-08:
I have tried push ESP3D button and opera and ms edge opened https://192.168.0.1 blank side. I typed http://192.168.0.1/ in browser and this don`t help, no changes still opening https://192.168.0.1 blank side.

luc-github, 2023-03-08:
I do not understand how you can open `http://192.168.2.1?forcefallback=yes` but opening `http://192.168.0.1` open http**s**://192.168.0.1,  that is weird but it is not linked to ESP3D Firmware - it is in your opera or you antivirus/firewall or OS

even if you enabled `Always use secure connections`  in opera you should be able to go to http , I have tested on my opera and it works

[image]

szczarek, 2023-03-08:
I turn off firewall, antyvirus and options in opera that you told and that doesnt work. I tryied on android chrome and once I can go to ESP3D settings, I changed a settings to connect my router wifi but it doesent connects and now cant go to the settings, only http://192.168.0.1?forcefallback=yes. I flashed espcam one more time but it no helps.

luc-github, 2023-03-08:
and your page still redirect to https instead of http ?

szczarek, 2023-03-08:
Yes is going to https.

szczarek, 2023-03-08:
IT WORKS! thank you! It connects to router wifi.

closed 2023-03-08 by szczarek

luc-github, 2023-03-08:
well you should find better solution - it seems you have something redirecting simple address , check your browser extension , security settings, vpn as I see some vpn icon, the issue is not in esp3d