luc-github/ESP3D-WEBUI #347: listing files on TFT-SD sends M20 instead of accessing built-in SD card

3D-Darryl, 2023-03-28:
WebUI 3.0.0-a31.M2.1
FW Version 3.0.0.a222
Marlin 2.1.2
ESP32 AI-Thinker with camera and SD card.  32GB SD formatted FAT32.
Both SD and TFT-SD enabled.  Global file system enabled.
Selecting "SD" produces 8.3 format file name list from the printer's SD.
Selecting "TFT SD" sends "M20" to Marlin and ignores the result.
Additionally there is no "Upload" selection for either SD card.

closed 2023-04-23 by luc-github

3D-Darryl, 2023-04-24:
I do not know how to answer your question"What TFT are you using"
AI Thinker has ESP32S with built in camera.
I do not get notification until you close the issue.

3D-Darryl, 2023-04-29:
I replied to the email which apparently was ignored.  These issues still exist:
Hi, just tried the latest version UI:3.0.0-a31.M2.1, FW 3.0.0.a224 and saw the following issues:
1.  When I use the web page to upload a file to SD, the Windows file explorer shows a ".tmp" file, but the web page shows the .gcode file.  When I use file explorer to move a file, the web page shows a ".tmp" file while file explorer shows .gcode.  Refreshing the UI or the file explorer does not make a difference.
2. When I click on the PLAY button on SD  I get a message "open failed, File: cal.gcode."
3. When I select TFT and select REFRESH, the serial display shows the file names but they do not appear in the UI window.
It is looking better, thanks for all your efforts!

luc-github, 2023-04-29:
When Issue is closed I do check update, next time please open new issue and follow template 

TFT_SD need to be used when you have TFT - I am guessing it is not your case based on your comment but you try to use it list / print from your printer -> it won't work because commands are designed to BTT TFT SD card and Gcodes

if you have Direct SD on esp3d and want to use printer SD you must enable extra SD in settings of webUI

if you try to print from ESP3D direct SD card - it is not possible because printer does not have access to it,  that is why you have `open failed, File: cal.gcode.`

Stream GCODE from ESP3D sdcard is under development