luc-github/ESP3D-WEBUI #350: [BUG] grblHAL WebUI3.0 Status not updating when streaming, no resume button, more gui based jogging

troth530, 2023-04-07:
**Describe the bug**
Thank you for your excellent efforts!
1) When streaming through SD card the status does not update.
2) There is a pause button but no resume button.
3) How to enable the more gui based jog controls, such as the circular jog controls? Attached shows the square boxes WebUI3.0.  
Thank you!

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
I believe these are WebUI GUI settings, I am not familiar with how to fix.

**Screenshots**
[image]
[image]

**ESP3D Firmware: **
- Version:[grblHAL]
- Web UI Version:[3.0.0-a.34.GH2]
- Wifi mode:[STA & AP]
- Any change done in FW

**Target Firmware: **
 - Name: GrblHAL
 - Version [1.1f 2022 Oct]

**Board used :**
 - MCU: [ESP32] 
 - Name:[esp32dev] 
 - Flash size: [e.g. 4M: 2M/2M]

**Browser used (please complete the following information):**
- Name: [Chrome]
- Version  [111]
- Operating system [Windows 10]

**Additional context**
Thank you for your excellent efforts!
1) When streaming through SD card the status does not update.
2) There is a pause button but no resume button.
3) How to enable the more gui based jog controls, such as the circular jog controls? Attached shows the square boxes WebUI3.0.  
Thank you!

luc-github, 2023-04-07:
you need to add `?` in polling commands or you won't have any status update.

resume button will appear according status update

The missing jog controls will appear when the first `?` will be received and so automatically detect the axis numbers

troth530, 2023-04-08:
> you need to add `?` in polling commands or you won't have any status update.
> 
> resume button will appear according status update
> 
> The missing jog controls will appear when the first `?` will be received and so automatically detect the axis numbers

Hi Luc, thank you so much.  The ? in polling indeed fixed the status update and the pause button becomes a resume button now as well.  

However the jogging controls have more information, but what I was hoping for was as below with the circular jog controls 
[image]

Also I am unable to load the themes.  I upload to SPIFFS and they are detected, but at the dashboard upon reloading it will say "File not found theme-xxx"

Thank you!

luc-github, 2023-04-08:
Ho you mean the pronterface UI, this one  is in not used in grb_hal/ grbl only in 3D printer webUI, cnc user told me they prefer simple buttons UI  so pronterface is not availalble  on cnc web ui

About the theme issue it may be a bug indeed as on grbl path is different - I will have a look

luc-github, 2023-04-08:
I cannot duplicate can you do a screenshot of you filesystem panel 
also can you give the output of 
`[ESP800]version=3.0.0-a34` command ?

troth530, 2023-04-08:
Hi thank you for looking into this...
[image]

[ESP800]version=3.0.0-a34
{
 "cmd": "800",
 "status": "ok",
 "data": {
  "FWVersion": "1.1f",
  "FWTarget": "grblHAL",
  "FWTargetID": "80",
  "Setup": "Enabled",
  "SDConnection": "direct",
  "SerialProtocol": "Socket",
  "Authentication": "Disabled",
  "WebCommunication": "Synchronous",
  "WebSocketIP": "192.168.1.177",
  "WebSocketSubProtocol": "webui-v3",
  "WebSocketPort": "81",
  "Hostname": "grbl",
  "WiFiMode": "AP",
  "FlashFileSystem": "littlefs",
  "HostPath": "/",
  "WebUpdate": "Disabled",
  "FileSystem": "flash",
  "Time": "Manual",
  "Axisletters": "XYZA"

luc-github, 2023-04-08:
Ok and now can you share a screenshot of  what theme you  did choose in settings/interfaces and a screenshot of the error message

luc-github, 2023-04-08:
also I do not see the capacity/occupation  in FS - is it missing ?

troth530, 2023-04-08:
> Ok and now can you share a screenshot of what theme you did choose in settings/interfaces and a screenshot of the error message

[image]

> also I do not see the capacity/occupation in FS - is it missing ?

[image]

Here's a video if it helps.  Thank you!
[https://drive.google.com/file/d/10BJRJWBzqUUoxP_hskW6ItbB7ocK3sK2/view?usp=share_link](https://drive.google.com/file/d/10BJRJWBzqUUoxP_hskW6ItbB7ocK3sK2/view?usp=share_link)

luc-github, 2023-04-09:
I will setup a grblHAL system today to check

luc-github, 2023-04-09:
The issue is not in webui but related to grblHAL - I have opened issue there

luc-github, 2023-04-10:
On ESP3D firmware with webui without grblHAL fix - everything is work as expected 
[image]
 
But with latest version I have time out connection on ESP3D so I will review what is wrong tomorrow

closed 2023-04-13 by luc-github