luc-github/ESP3D-WEBUI #351: the movement of the axes is reversed compared to the GRBL configuration[BUG]

Ulisse76, 2023-04-15:
In the jog rose the movement of the axes is reversed compared to the GRBL configuration.
Movement works fine under Lightburn/LaserGRBL and TS35 monitor.

luc-github, 2023-04-15:
what version are you using ?which what board?

Ulisse76, 2023-04-15:
firmware: 2.10 
board: MkS DLC32 2.1

luc-github, 2023-04-15:
you mean when you ask X+10 it is not doing X+10?

Ulisse76, 2023-04-16:
when I ask for X10 it seems to send to controller -X10.
Can you show me the point in the HTML code where I could manually reverse this thing?

luc-github, 2023-04-16:
the code is minified - it is not that simple - also your issue is weird 
 
Can you do a screenshot of what you are doing ?

I cannot duplicate your issue when pressing -X side I got -X and when pressing +X side I have +X
[image]

Ulisse76, 2023-04-16:
yes, i have seen the GRBL commands and they are being fixed.

  At this point I think it's a Firmware problem (all versions).
Firmware does not invert the axes when it takes commands from WebUI or APP (therefore from WebSocket).

The problem arises from the fact that in some CNCs it is necessary to invert the movement of the axes from the firmware --> X/Y direction pin invert ($3)

Can you help me identify in the code the point where I could put two radiobuttons to reverse X and Y ?

luc-github, 2023-04-16:
I do not understand - you wrote commands are incorrect and now you mention commands are corrects
 also now you write it is FW issue and you want to fix it  in webui ? what do you expect that adding radio buttons will do ?

Ulisse76, 2023-04-16:
yes, it's only been a few hours since I've been working on the problem and I'm slowly realizing where the problem is, looking for a solution.
I can't figure out if it's a firmware or WebUI issue.

Adding the two radiobuttons in my opinion is a suggestion to improve the functions of the page.
if you prefer you can close the thread and edit the Html by yourself as I did in the RoseJog thread.

Thank you

luc-github, 2023-04-16:
Sorry - my issue is more : `I do not understand your issue` because what you describe is not clear neither consistent 
So I cannot fix an issue when  I do not understand what is the problem you meet

luc-github, 2023-04-16:
I did this - check settings to adjust 
[index.html.gz](https://github.com/luc-github/ESP3D-WEBUI/files/11241835/index.html.gz)

Let me know if this address your problem ?

Ulisse76, 2023-04-17:
I identified (and solved) the problem better.
Unfortunately Lightburn, depending on the position of the homing, reverses the DIRECTION of the axes. 
Playing with the GRBL parameters I put the movements back in the right direction.
thank you for the support then and if you want we can delete the thread.

luc-github, 2023-04-17:
indeed there was a typo and I did not tested fully 
please give a try to this one - it should help to reflect real cnc orientation
[index.html.gz](https://github.com/luc-github/ESP3D-WEBUI/files/11248582/index.html.gz)

Ulisse76, 2023-04-17:
now it's OK!
 it would be nice to have two radio buttons to reverse the directions at will.

Know that this change is actually important because I haven't solved the problem with the Homing position issue.

luc-github, 2023-04-17:
>it would be nice to have two radio buttons to reverse the directions at will.

What do you mean ?

There are reverse X and reverse Y settings - what your settings should do in addition of existing one ?

closed 2023-04-21 by luc-github