luc-github/ESP3D-WEBUI #353: [BUG] Serial communication issue

dryas, 2023-04-18:
**Describe the bug**
When I try to access the machine information I get an error message. When I try to run M114 I simply get an "ok" in the terminal. The position of XYZ are not updated automatically and as the response of M114 is only "ok" I also can't get it to work with automatic polling. On the other side M105 seems to work. M503 is also only returning an "ok". It seems like there is some issue in the serial communication (probably on line breaks?) or the parsing engine. Additional: M115 returns only one line, M119 also only an "ok" and no endstop info (see screenshot).

**To Reproduce**
Steps to reproduce the behavior:
Run M114 in the terminal, run M503 in the terminal.
Response is only "ok", but no more data.

**Expected behavior**
M114 returns the position of the extruder
M503 returns the settings of the printer
I can use M114 to update the XYZ positons on the jog panel.

**Screenshots**
[image]

**ESP3D Firmware:  (please complete the following information)**
- Version:3.0.0.a225
- Web UI Version: 3.0.0-a35.Me2
- Wifi mode: Client Mode
- Any change done in FW: none

**Target Firmware:  (please complete the following information)**
 - Name: Marlin
 - Version 2.1.2

**Board used (please complete the following information):**
 - MCU: ESP8266
 - Name: MKS Wifi 
 - Flash size: 4M

**Browser used (please complete the following information):**
- Name: Chrome
- Version  ?
- Operating system MacOS 12.x

**Additional context**
Few questions on that:
- Should the temperature and Jog position info be automatically updated? When I activate the "Automatic data update" option in the settings nothing happens. I need to use the polling function to update the temperature (and position if it works) at the moment. Is this the right way to go?
- Not related to that, but as I didn't found an answer: Whats the difference between "Marlin" and "Marlin_Embedded" on the index.tar.gz files?

luc-github, 2023-04-18:
did you enabled the polling commands M114;M105 in settings ?

dryas, 2023-04-18:
In the Marlin or in the ESP3D settings? They are activated in Marlin. I haven't seen them in the ESP3D config. Let me have a look.

luc-github, 2023-04-18:
in webui interface tab

dryas, 2023-04-18:
Like this? Yes, they are added as "polling" commands. Temperature works, but not position (M114). 
[image]

luc-github, 2023-04-18:
you  can add like this or to existing one just separate by `;`

luc-github, 2023-04-18:
another solution is to enable auto report in Marlin so no need to do polling

dryas, 2023-04-18:
Ah, good to know. Then I will probably combinate them later on. However, M114 does not update the positions on the Jog control and for example when I open the Machine tab I only get an error (but I'm not sure if this is already implemented in to the WebUI3.0).

Regards the auto report: Will it work with SD card uploads? I use the MKS protocol and I've read somewhere that this could be result in complications and interferences when commands get mixed in the communication.

[image]

luc-github, 2023-04-19:
Yes it is implemented 
but may be not enabled in Marlin
What is the result of `M503` in webui terminal ?

luc-github, 2023-04-19:
on MKS protocol yes it may be safer to not use autoreport

luc-github, 2023-04-19:
if you use MKS protocol - I am guessing `M503` will report nothing https://github.com/luc-github/ESP3D/discussions/620
Unless they fix the issue in Marlin

luc-github, 2023-04-21:
@dryas   issue be closed ?

dryas, 2023-04-21:
@luc-github Yeah, I've investigated a little bit more and it seems like MKS has encapsulated the communication in their logic somehow. I've moved back to the traditional serial protocol without the possibilities to upload.

Thanks for all your hard work. You could be proud of what you have created!

closed 2023-04-21 by luc-github