luc-github/ESP3D-WEBUI #358: Selecting TFT SD on the UI sends "M20: to Marlin but ignores the result

3D-Darryl, 2023-04-29:
**Describe the bug**
Selecting "TFT SD" in the UI causes Marlin to list the files on the Printer's SD but the UI ignores the response

**To Reproduce**
Select "TFT SD".  Select "refresh" if needed

**Expected behavior**
The file list that Marlin sent should appear in the FILES window and GCODE files should have  a PLAY button.

**ESP3D Firmware:  (please complete the following information)**
WebUI 3.0.0-a31.M2.1
FW Version 3.0.0.a225

**Target Firmware:  (please complete the following information)**
Marlin 2.1.2

**Board used (please complete the following information):**
ESP32 AI-Thinker with camera and SD card. 32GB SD formatted FAT32.

**Browser used (please complete the following information):**
Firefox
Windows 10

**Additional context**
Flash and SD and TFT-SD enabled. Global file system enabled.

luc-github, 2023-04-29:
as explained before TFT SD is not printer SD - https://github.com/luc-github/ESP3D-WEBUI/issues/347#issuecomment-1528639345
the TFT M20 reponse format is different than Marlin M20 response format so Marlin response will be ignored because you do not have TFT and TFT won't answer instead of Marlin

closed 2023-04-29 by luc-github

3D-Darryl, 2023-04-29:
Your terminology is confusing: "TFT-SD" is the SD card on the ESP32 board or is it the SD card on the Printer?  
The Printer's SD card contents should show up in the UI as 8.3 filenames,  This does not work (it worked in a previous revision).
The UI should be able to start printing from the printer's SD card, but this is not possible because the file list is broken.

luc-github, 2023-04-29:
The TFT-SD is the SD on TFT screen this is not confusing sorry

3D-Darryl, 2023-04-30:
My Ai Thinker ESP32 does not have a screen.  My printer has a LCD screen.  which one is "TFT SD"?

luc-github, 2023-04-30:
obviously you do not have any TFT screen if you have you would knonw it, you only have a LCD screen ,more over I gave you a link that you seems not checking showing the TFT screen example

3D-Darryl, 2023-04-30:
OK, checked out the link, now I see where you are coming from.  Thanks

luc-github, 2023-04-30:
and TFT-SD  can only be used if ESP board is connected to TFT screen

3D-Darryl, 2023-04-30:
My Marlin printer with LCD screen has an SD card that can be accessed by standard Gcode commands.  I expect this is the equivalent of your "TFT-SD"

luc-github, 2023-04-30:
ho boy did you check the link I gave you ? 
is your screen identical ? 
is it a touch screen ? 
Is it a  BTT touchscreen ? 
it using  Serial to communicate with your printer ?

L

3D-Darryl, 2023-04-30:
All that is irrelevant.  There is an SD card connected to the Mega2560 that has the GCODE files.  Serial commands from the ESP32 can send GCODE commands to read the file list and start/stop printing.  What sort of screen it has does not matter.
Not that it makes any difference but here is the screen:
https://reprap.org/wiki/File:RRD_FULL_GRAPHIC_SMART_CONTROLER_STARTUP.JPG

luc-github, 2023-04-30:
this is not TFT-Screen so SD on it is printer SD so TFT-SD command won't work 

Yes screen matter  because of SD and command to query this SD and commands to print

TFT use different command and their answer are different than printer SD as I already explained my answer to you : https://github.com/luc-github/ESP3D-WEBUI/issues/347#issuecomment-1528639345

luc-github, 2023-04-30:
If you considere all that irrelevant - I have no other answer sorry - it seems you know better than me of what I am doing and hardware I support - so I am useless for you sorry

3D-Darryl, 2023-04-30:
Please don't get hung up on hardware differences. I do not know how your system is set up or what capabilities it has.  I found your display on Amazon, there is more useful information there, but I don't understand where you have ESP32 connected to your system.
You had a past version of the UI that was able to display the files list from my printer's SD card using M23 and M24.  It did not show the "PLAY" button.  The most recent version does not work anymore (no file list).
My ESP32 has an SD card.  Your UI is able to display the file list on that SD card, and I can access it from Windows as well.  I understand the streaming from that SD card is not implemented yet.

luc-github, 2023-04-30:
because you choose wrong - you are the one who focus on TFT-SD nwhen you actually do not have it but keep saying it is the one|

I already told you here - https://github.com/luc-github/ESP3D-WEBUI/issues/347#issuecomment-1528639345
to enable Extra-SD in settings as you enabled direct SD in ESP32Cam - you do not read what I wrote and keep talking about TFT-SD

[image]

I am locking thread now