luc-github/ESP3D-WEBUI #362: [FEATURE REQUEST]

scorninpc, 2023-05-28:
**Is your feature request related to a problem? Please describe.**
My feature request is more about enhancement, create by third parties

**Describe the solution you'd like**
I was working on totaly javascript plugin. I found javascript methods to send command, create hooks, etc, and done a script that can create a progress bar with progress of printing

[image]

After see this working, I have idea that web developers can help add plugins to ESP3D just writing a regular javascript, without recompile/write esp/test loop

With this method, i see that we can completely rewrite some blocks, add new blocks, add new features to ESP3D

My idea is that will be nice if we have a way to add external links to some plugins (js files), and the interface load this scripts. Just online plugins for now

**Describe alternatives you've considered**
For now, I run my script on console. But a nice way is use grease monkey web extension to run this scripts

**Additional context**
I can write and host a documentation about ESP3D javascript methods, or can write a  javascript object to make this easer. Also, i can write and host a plugin repository

luc-github, 2023-05-28:
what is the difference with existing extension feature implemented in ESP3D V3 ?
https://github.com/luc-github/ESP3D-WEBUI/blob/3.0/extensions_samples/API.md

scorninpc, 2023-05-28:
> what is the difference with existing extension feature implemented in ESP3D V3 ? https://github.com/luc-github/ESP3D-WEBUI/blob/3.0/extensions_samples/API.md

wow, this is nice. I'm using V2, but this API will help a lot

How to load this extensions on V3?

luc-github, 2023-05-28:
V3 has modulars block for dashboards, just upload extension file on local FS and add a block as extension in webui settings:
https://esp3d.io/esp3d-webui/v3.x/documentation/index.html

Documentation is not yet completed but main is there

Yes it usable - still some features to implement but it is already used by several people.
But be noted it need ESP3D V3 or ESP3DLib V3 - it is not compatible with ESP3D v2.x neither ESP3DLib V1 and won't be back ported

scorninpc, 2023-05-28:
Thank you @luc-github , this is exactly what i'm thinking. I'll upload V3 to see this today.

closed 2023-05-28 by scorninpc

luc-github, 2023-05-28:
if you want write more extension examples or documentation feel free