luc-github/ESP3D-WEBUI #364: Allow sending blank lines from the MDI (Terminal).

terjeio, 2023-08-16:
**Is your feature request related to a problem? Please describe.**
grblHAL repeats the last error until cleared by a blank line or a $-command for safety reasons. The v3 WebUI does not allow sending blank lines from the MDI.

**Describe the solution you'd like**
Allow sending blank lines.

**Describe alternatives you've considered**
Setting [grblHAL compatibility level](https://github.com/grblHAL/core/wiki/Compatibility-level) > 0 disables this safety feature alongside some other new features supported by the WebUI - tool changes is at least one prominent one. IMO not a good solution.

**Additional context**
See this [OpenBuilds comment](https://openbuilds.com/threads/error-2s-new-blackbox-x32.20746/#post-137256) onwards.

luc-github, 2023-08-16:
what is a blank line ?
`\n` ?
"" ?

terjeio, 2023-08-16:
`http://192.168.2.186/command?cmd=` works after I modified my backend to test for blank lines. I just forward a newline to the command parser when received.

luc-github, 2023-08-16:
I have modified the webui to allow nothing as test but I forgot it add a timestamp to bypass the cache :
` http://192.168.2.186/command?cmd=&t=1692164825367` does this still work on your backend ?

terjeio, 2023-08-16:
Timestamp is ok and I guess needed when there is a proxy in between.

luc-github, 2023-08-16:
Can you give a try to :
[index.html.gz](https://github.com/luc-github/ESP3D-WEBUI/files/12356237/index.html.gz)

terjeio, 2023-08-16:
Works! Thanks for quick implementation.

luc-github, 2023-08-16:
Ok I will push it - I have just changed one thing compare to the version you test : emptyline won't go to history of command because it is not easy to see what command is that

luc-github, 2023-08-16:
@terjeio you are welcome , it was an easy fix and happy it is useful ^_^

closed 2023-08-16 by luc-github