luc-github/ESP3D-WEBUI #371: [FEATURE REQUEST] webcam, iOS/android app?

Wirepower, 2023-11-09:
Ability to have webcam stream added
And also a iOS/android app to link to webgui interface?

luc-github, 2023-11-09:
I do not understand webui 2.1 and 3.0 already support ip webcam panel, and 3.0 also support esp33cam control

can you elaborate?

Wirepower, 2023-11-09:
Oh ok, some reason mine doesn’t show anything for web cam interface. Might need to check what version I have.

Wirepower, 2023-11-10:
So this is for the ESP-01S.
Maybe that’s why no camera function? 
I was hoping to have a webcam via web stream. So I can broadcast my rpi webcam on the webgui.

luc-github, 2023-11-10:
what esp3d version have you ?

Wirepower, 2023-11-10:
It’s 3.0 webgui and firmware

luc-github, 2023-11-10:
the doc is pretty explicit IMHO: https://esp3d.io/esp3d-webui/v3.x/documentation/camera/index.html
Why do you say you do not see how to add a stream ?

Wirepower, 2023-11-10:
I’m new to this so, bear with me. I will have a look. I thought it would have been a seperate panel and not under extra features so I may of missed it

luc-github, 2023-11-10:
it will create new panel or page according your needs

Wirepower, 2023-11-10:
Ok thank you so much. I found it. However is there a way I can flip the image?

luc-github, 2023-11-10:
Sorry the panel only display the webpage - it does not control the remote page - you may have the command on your remote page to control the output, some camera allow to add command in url