luc-github/ESP3D-WEBUI #376: [FEATURE REQUEST] Store terminal history in local storage

dbuezas, 2023-12-03:
**Is your feature request related to a problem? Please describe.**
I miss a history like in the shell

**Describe the solution you'd like**
That the history is stored in local storage so I can arrow up even after reloading the page. Ideally an extra popup that shows the history hidden under a button in the down caret

Here:
[image]

**Additional context**
I think I'll give this a try myself in a couple of days :)

luc-github, 2023-12-04:
Yes sorry all localstorage request are no my priority currently - also be sure it won't increase the final footprint package more than few bytes I want to keep the final size as small as possible

luc-github, 2023-12-04:
for localstorage you should not need any for sure

luc-github, 2023-12-23:
I close issue as fixed then

PR #378 (2023-12-05): Persistent terminal history
The feature costs 106 bytes, the bundle grew 0.1%.

```
-rw-r--r--@   1 davidbuezas  staff   90922 Dec  4 14:17 index.html.gz.after
-rw-r--r--    1 davidbuezas  staff   90816 Dec  4 14:28 index.html.gz.before
```

https://github.com/luc-github/ESP3D-WEBUI/assets/777196/80d6ba3e-c2e8-4902-b9de-7f87aed2da2c

Closes #376
closed 2023-12-23 by luc-github