luc-github/ESP3D-WEBUI #379: [Question]WPOS and MPOS on ABC axis

luc-github, 2024-01-11:
@terjeio I am still  in learning curve for grblHAL to do better webUI and TFT UI implementation
My understanding is that A/B/C axis are rotation axis - in that case does the concept of WPos and MPos has any meaning ?
Do it mean something to display WPos and Mpos for this axis - only Mpos is relevant or am I wrong ?

As 6 axis grbl guru could you confirm / correct my understanding ?

terjeio, 2024-01-12:
I just display the rotary axes as any other in my sender, I do not see any reason to handle them differently other than perhaps change the unit of measure (if displayed). Note that grblHAL per default does not designate any axis as rotational per default, setting `$376` is used for that - it is a bitmap where bit 0 is A, bit 1 is B up to bit 4 which is V.
Note that designating an axis as rotational only disables the inches to mm conversion internally, and changes the settings units reported for the axis.

luc-github, 2024-01-12:
Hi, thank you for your feedback - yes from programming point of view somehow all axis have same characteristics I fully agree

But I try to understand pratical situations, after the 3 dimensions : X Y Z,  what kind of movement an Axis is doing ? and only rotation came to my understanding (spatial geometry is not my best skill...)
And so I was wondering if displaying WPos for these raxis make sense if they are rotation.
As you on WebUI, i  display all because there is space for all 2x6 positions (WPos/Mpos)

My question is due to small rendering, I am working on a TFT project for grbl / grblHAL (https://github.com/luc-github/ESP3D-TFT) and on such screen space is limited but still need to be user friendly.

when most of users use 3~4 axis, which is not an issue, displaying 2x 6 axis positions become a challenge on  some screens , so I was wondering if I can optimize the display of the rotational axis (A,.B,...)  by removing WPos coordinate if they have no meaning on them

terjeio, 2024-01-13:
> what kind of movement an Axis is doing ?

It could be linear or rotational depending on the $376 setting, IMO it is best to think of the motion in terms of how the tooltip moves relative to the workpiece.

> And so I was wondering if displaying WPos for these raxis make sense if they are rotation.

I do not know, I have never used a 4-axis machine. And for my 3-axis machines I do not care about WPos since I do not home. Perhaps someone using a rotary can chime in?

luc-github, 2024-01-15:
@terjeio thank you a lot for sharing  
>for my 3-axis machines I do not care about WPos since I do not home

That a good point as most of people will do a manual 0 according the job and the object they are working on

>I have never used a 4-axis machine.

I also think 4 axis is not so common, and 5/6 axis even fewer people use them, so I am wondering if I should support them on TFT and just keep the current support on WebUI as it is

Thank you again for your valuable feedback ^_^

closed 2024-01-15 by luc-github