luc-github/ESP3D-WEBUI #386: [FEATURE REQUEST] Sort Gcode files by "date modified" or leave them unsorted as provided by host/TFT

rondlh, 2024-04-22:
**Is your feature request related to a problem? Please describe.**
Typically I want to print the last file I copied to the USB flash drive in my TFT display when I issue a M20 U: (list USB files).
Unfortunate ESP3D sorts the files alphabetically, so I have to dig around to find the file I want to print. 

**Describe the solution you'd like**
Add an option to 1. sort the files alphabetically, or 2. don't sort them because the TFT handles this already.

**Describe alternatives you've considered**
This can also be hard coded, then leave it up to Marlin or in my case the TFT to provide the files in the required order (which can be configured).

**Additional context**
I use ESP3D V3.0, which I can recommend over V2.1

kleffa, 2024-05-07:
https://marlinfw.org/docs/gcode/M020.html
well only in latest Marlin are avaible the time stamp for M20 , so I think they can be sort by date or name

kleffa, 2024-06-05:
ESP3D didn't sort files, they are just in order provided, which is a Marlin feature, maybe you can try make a better UI to make god lists or change options in marlin to sort them by date

rondlh, 2024-06-05:
ESP3D sorts the files in src\components\Helpers\filters.js
This has nothing to do with Marlin, because the M20 response is provided by the TFT, not Marlin.

luc-github, 2024-07-02:
Fixed by https://github.com/luc-github/ESP3D-WEBUI/commit/423dbae646ada07fe3088f0e66b46ab671b21491

closed 2024-07-02 by luc-github