luc-github/ESP3D-WEBUI #387: What is the function of the "Play file" button in the SD card panel?

Mitko125, 2024-05-01:
Samples were made on:
ESP3D VERSION 3.X
ESP3D-TFT VERSION 1.X
ESP3DLIB VERSION 3.X with https://github.com/luc-github/Marlin/tree/ESP3D-V3-2.1.x
I am using ESP3D-WEBUI VERSION 3.X

The specific example is for:
FW version: 2.1.2-3.0.0.a222
FW target: marlin
ESP3D-WEBUI: https://github.com/luc-github/ESP3D-WEBUI/tree/3.0/dist/Printer3D/Marlin

Result obtained:
"
M21
M23 /K3D_RCT_H190-B60_1.2-1.2mm_10-70mms.gcode
M24
echo:SD card ok
ok
echo:Now fresh file: /K3D_RCT_H190-B60_1.2-1.2mm_10-70mms.gcode
open failed, File: K3D_RCT_H190-B60_1.2-1.2mm_10-70mms.gcode.
ok
"

I can read and write programs on the ESP3D SD card. But I can't run a program to print on the printer and occasionally monitor the work over Wi-Fi as well.
I can't even save a program to the printer's SD card via wifi and then run it from the printer.

Total failure so far. ESP3D cannot run the printer program from its SD card. I can only move the printer axes via wifi. Is that the goal?

So far I'm only getting responses like "this doesn't work...".
Huge respect for the great work you have put in. But explain to me how I can get my printer to run by program and monitor it via wifi. Please tell me where I am doing wrong. In what device should I put ESP3D to get this functionality or is it impossible.

The installations are prepared and if necessary I can turn them on and answer questions and do experiments.

Or tell me that such functionality will not exist and I will not bother you:
"I want on the SD card of some ESP3D device to save and run a program for the printer via wifi."

luc-github, 2024-05-01:
The response is - it depend on ESP3D version (2.x vs 3.0),  Firmware target (Marlin / grbl) and hardware connection with SD reader (Direct / Shared / None)

For 2.x: There is not direct SD connection suppport, only System SD on 3D Printer, no SD Support for grbl, Monitoring is working, SD Upload using M28/M29 is implemented but unfortunatly it is often broken so support is dropped as explained here: http://esp3d.io/esp3d/v2.x/documentation/sdtransfer/index.html

For 3.0: This  is explained here : http://esp3d.io/esp3d/v3.x/documentation/sdtransfer/index.html
and here  https://github.com/luc-github/ESP3D/discussions/540#discussioncomment-2897564, monitoring is working

your example :
```
M21
M23 /K3D_RCT_H190-B60_1.2-1.2mm_10-70mms.gcode
M24
echo:SD card ok
ok
echo:Now fresh file: /K3D_RCT_H190-B60_1.2-1.2mm_10-70mms.gcode
open failed, File: K3D_RCT_H190-B60_1.2-1.2mm_10-70mms.gcode.
ok
```
Should start the print if file is on printer SD and Marlin is configured for long name 

Uploading on SD depend on hardware used as mentionned here: http://esp3d.io/esp3d/v3.x/documentation/sdtransfer/index.html

luc-github, 2024-05-01:
Function like gcode streaming by ESP3D from it's dedicated SD card is currently experimental in ESP3D-TFT firmware 1.0 and only for Marlin based on ping-pong streaming for 3D printer commands (command / ack, command/ack, etc...), grbl version of Marlin is not suppported 
http://esp3d.io/index.html has some flow to explain what is work and what to use

Mitko125, 2024-05-01:
If you have read my letter, I have specified exactly which device the example is with.
It's a joke that with this device I uploaded a program to the SD card via wifi with ESP3D -WEBU. Then I ran this program through UART with pronterface.exe and it ran without a problem.

I repeat the question:
Why can't I start the program with the "Play file" button, where is it described, what settings to make in ESP3D -WEBU. Because with pronterface.exe through UART it apparently works. I don't know (there is no example) how to do with the "Play file" button.
Forgive my stupidity, but I didn't see anywhere that it says what setting I'm missing.

luc-github, 2024-05-01:
> I have specified exactly which device the example is with.

Sorry I do not see what main board you are using, neither what ESP board you are using , where your SD is connected to,  neither how you connected esp board to you system board

you only describe the version but the version you mentioned : FW version: 2.1.2-3.0.0.a222 is actually Marlin using ESP3DLib which is not ESP3D 

Can you clarify ?

Mitko125, 2024-05-01:
I use ESP3D as a generic name.
I have failed with ESP3D, ESP3DLib and ESP3D-TFT.
I've used all of them with ESP3D-WEBUI: https://github.com/luc-github/ESP3D-WEBUI/tree/3.0/dist/Printer3D/Marlin
The board is implemented with ESP3DLIB VERSION 3.X with https://github.com/luc-github/Marlin/tree/ESP3D-V3-2.1.x.
The board itself is a derivative of BOARD_MRR_ESPA.
With simulators of #define TEMP_SENSOR_0 999 and #define TEMP_SENSOR_BED 998
He can't go home either, but that doesn't matter. But it runs a program launched through UART with pronterface.exe. (I saved the program to the SD card through ESP3D-WEBUI)

When the ESP32 memory is deleted, a wifi connection is set. Index.html.gz is installed. Enabled SD card in settings.
Unfortunately "Play file" does not work.
What haven't I done right?
I am working according to: https://esp3d.io/esp3d-webui/v3.x/index.html
Maybe he should read elsewhere?

I'm repeating myself:
The printer works, but I can't launch a program via wifi.

luc-github, 2024-05-01:
>I use ESP3D as a generic name.

Well that is not helping to clarify things 

If you use BOARD_MRR_ESPA then your esp32 is the main and only board - where Marlin is installed and ESP3DLib is enabled 
Then the index.html.gz to be used is `https://github.com/luc-github/ESP3D-WEBUI/tree/3.0/dist/Printer3D/Marlin-embedded` not the Marlin one.
be sure you enabled SD card in Marlin and WEBUI can see it 
If failed like the message you shared above, try to use short name for the file because even Marlin says long name is supported it failed sometimes

To sum up On ESP32 board flashed as BOARD_MRR_ESPA  with   https://github.com/luc-github/Marlin/tree/ESP3D-V3-2.1.x upload , play file, monitoring should work

Mitko125, 2024-05-01:
I am ecstatic. I renamed the file to "Test.g" and the stepper motors sang.
The key is in the short names (type DOS 8 name 3 extension).
Now I guess also with your other tools (ESP3D, ESP3D-TFT).

ESP3D-TFT I will use it with GRBL (I have a lot of experience with ESP-IDF and LVGL).

Somewhere in capital letters note short names.

Thank you very much for your time and excuse me for the insistence.

luc-github, 2024-05-01:
>I renamed the file to "Test.g" and the stepper motors sang.

That is Marlin bug not ESP3DLib neither WebUI

>ESP3D-TFT I will use it with GRBL (I have a lot of experience with ESP-IDF and LVGL).

There is not support yet - I am working on it

>Thank you very much for your time and excuse me for the insistence.

Using clear informations / description  help to solve issues ;)

closed 2024-05-02 by luc-github

luc-github, 2024-05-02:
@Mitko125 if you have esp32 as main board for grbl you may consider to use : https://github.com/grblHAL/ESP32 it also use ESP3D-WEBUI 3.0 for grblHAL : https://github.com/luc-github/ESP3D-WEBUI/tree/3.0/dist/CNC/GRBLHal
it support SD natively

Mitko125, 2024-05-02:
Thank you very much for the information, I am aware.
I have a CNC3018 and I am preparing a laser engraver. But they are with Arduino UNO derivative boards. So far I've been running them from a computer.
For them, ESP3D-TFT is of great interest (we forget about the computer).
So I will work on that.

I also have a real 3D printer (not a simulator). I will connect it to ESP3D (hope I don't have to trouble you).

Many thanks for your responsiveness. I wish you success in everything you do!!!