luc-github/ESP3D-WEBUI #388: [BUG]Extra SD have shows no folders

kleffa, 2024-05-06:
**Describe the bug**
when using M20L fileexplorer doesn´t show folders

**To Reproduce**
1 this is with M20
[image]
2 change SD list comand to M21;M20L
3 see Extra SD on files
[image]
the folder "/" shows just a empty folder, didn´t touch more 
also M23 in Marlin doesn´t work officialy with long names

**Expected behavior**
See files ordered by folders 

**Screenshots**

**ESP3D Firmware:  (please complete the following information)**
- Version: 3.0beta 3
- Web UI Version: 3.0.0a233
- Wifi mode: client
- Any change done in FW Camera enabled

**Target Firmware:  (please complete the following information)**
 - Name: Marlin 2.1.2.2

**Browser used (please complete the following information):**
- Name: Chrome  124.0.6367.119 (Build oficial)  and firefox 125.0.3

luc-github, 2024-05-06:
Can you provide the `M20 L` complete output ?

kleffa, 2024-05-06:
I have two folders in there, "04" and "08", M20L just put all files together, so can´t navigate folders to keep order, here the M21;M20L ouput 

M21

M20L

echo:SD card ok

ok

Begin file list

CE3E04~1.GCO 13164510 CE3e04-30x5_huesitos-4h16m-23g.gcode

04/CE3-15~1.GCO 8958851 /CE3-15xgaleonataco-4h26m-24g.gcode

04/CE3-2X~1.GCO 2096208 /CE3-2x Pirata-0h56m-4g.gcode

04/CE3-3X~1.GCO 17631963 /CE3-3xGoro_R-15h42m-95g.gcode

04/CE3-50~1.GCO 5262191 /CE3-50_huesitos-1h22m-7g.gcode

04/CE3-5_~1.GCO 4665811 /CE3-5_arakar-1h25m-9g.gcode

04/CE3-5_~2.GCO 4783962 /CE3-5_chilesaurio-1h12m-8g.gcode

04/CE3-5_~3.GCO 4666366 /CE3-5_domeikodactilo-1h11m-8g.gcode

04/CE3-5_~4.GCO 6233093 /CE3-5_estegouros-1h23m-9g.gcode

04/CE9A0D~1.GCO 3765417 /CE3-5_ictiosaurio-1h07m-8g.gcode

04/CE3-BA~1.GCO 13560180 /CE3-Barco_6xGaleon-7h19m-40g.gcode

04/CE3-BA~2.GCO 6239102 /CE3-Barco_6xPirata-2h49m-13g.gcode

04/CE3-GO~1.GCO 5246172 /CE3-Goroshogi_1xRecta-4h38m.gcode

04/CE3-GO~2.GCO 10426693 /CE3-Goroshogi_2xR-9h09m-55g.gcode

04/CE3-GO~3.GCO 2779101 /CE3-Goroshogi_Exp_2xR-2h09m-13g.gcode

04/CE3-KO~1.GCO 19017814 /CE3-komikan_24xperro-5h22m-28g.gcode

04/CE3-KO~2.GCO 3078002 /CE3-komikan_2xPuma-0h55m-5g.gcode

04/CE3-LE~1.GCO 19002940 /CE3-Legion Barricada-4h34m.gcode

04/CE3-PI~1.GCO 18013935 /CE3-Piezas-7h38m-49g.gcode

04/CE3-QU~1.GCO 11065272 /CE3-quarto 6xCuadr-11h04m-79g.gcode

04/CE3-QU~2.GCO 11299200 /CE3-quarto 6xhexa-10h05m-68g.gcode

04/CE3-SH~1.GCO 13470718 /CE3-Shogi_1xR-11h05m-66g.gcode

04/CE3-SH~2.GCO 856218 /CE3-Shogi_Exp_2xR-0h41m-4g.gcode

04/CE3E04~1.GCO 20643043 /CE3e04-CALCIFER_cults-6h28m-42g.gcode

08/CE3E08~1.GCO 3702027 /CE3e08-arakar-1h16m-22g.gcode

08/CE3E08~2.GCO 3915118 /CE3e08-chilesaurio-1h06m-19g.gcode

08/CE3E08~3.GCO 3726164 /CE3e08-domeikodactilo-1h13m-20g.gcode

08/CE3E08~4.GCO 5044401 /CE3e08-estegouros-1h10m-23g.gcode

08/CE4F94~1.GCO 3205100 /CE3e08-ictiosaurio-1h07m-18g.gcode

CE3E04~2.GCO 23636275 CE3e04-33_arakar-8h32m-51g.gcode

CE3E04~3.GCO 25639657 CE3e04-33_chilesaurio-7h23m-45g.gcode

CE3E04~4.GCO 24009624 CE3e04-33_domeikodactilo-7h23m-44g.gcode

CE133C~1.GCO 32505132 CE3e04-33_estegouros-8h19m-51g.gcode

CEA9FC~1.GCO 20135376 CE3e04-33_ictiosaurio-6h57m-42g.gcode

CEF32E~1.GCO 10561759 CE3e04-088PA_L_MK1-4h06m-22g.gcode

CE19DC~1.GCO 0 CE3e04-CALCIFER_cults-6h28m-42g.gcode

CE410F~1.GCO 27910054 CE3e04-deathw-10h06m-60g.gcode

CE8CEE~1.GCO 4495695 CE3e04-Fermin_Serio-1h13m-8g.gcode

CE6490~1.GCO 4995684 CE3e04-Fermin_Serio-1h31m-11g.gcode

CE7131~1.GCO 5701488 CE3e04-Fermin_Serio-1h49m-11g.gcode

CE3E08~1.GCO 2128047 CE3e08-1quarto-2h12m-45g.gcode

CE3E08~2.GCO 2948251 CE3e08-60_huesitos-0h47m-11g.gcode

CE3-SH~1.GCO 13470734 CE3-Shogi_1xR-11h05m-66g.gcode

CED428~1.GCO 21291538 CE3e04-Laser mod S9-19h37m-123g.gcode

CE832C~1.GCO 3251596 CE3e04-mod screen-4h49m-36g.gcode

MOD1~1.GCO 5739231 mod 1.gcode

MOD2~1.GCO 27704227 mod 2.gcode

MOD3~1.GCO 18854798 mod 3.gcode

CE0043~1.GCO 2098116 CE3e04-Zero_Nose-0h41m-4g.gcode

CE3E08~3.GCO 307547 CE3e08-himobead_ressha-hiuma-0h20m-5g.gcode

CE3E08~4.GCO 2913438 CE3e08-audifonos 1-2h37m-55g.gcode

FANFA~1.GCO 172 fanfa.gcode

CE46F2~1.GCO 4853684 CE3e08-headband-side-right-5h42m-120g.gcode

CEF33F~1.GCO 4093068 CE3e08-pandita2-4h34m-94g.gcode

CE061B~1.GCO 6427563 CE3e08-panda2-left-5h01m-98g.gcode

CE56E2~1.GCO 923532 CE3e08-panda3-0h47m-12g.gcode

CE55BE~1.GCO 883597 CE3e08-firelordgary-1h24m-30g.gcode

CE4480~1.GCO 6320957 CE3e08-quarto cuadras 6-6h32m-133g.gcode

CEC2CD~1.GCO 6046295 CE3e08-quarto hexas 6-6h32m-133g.gcode

CE7E21~1.GCO 715102 CE3e08-Mount-0h58m-19g.gcode

08MOUN~1.GCO 715102 08mount 58m-19g - copia.gcode

End file list

ok

luc-github, 2024-05-06:
The path seems missing for long name :
e.g:
`04/CE3-SH~1.GCO 13470718 /CE3-Shogi_1xR-11h05m-66g.gcode` should be `04/CE3-SH~1.GCO 13470718 04/CE3-Shogi_1xR-11h05m-66g.gcode`

I will check with lastest release but looks  a bug in Marlin again for me may be because it is only number for folder name

Additionnaly per my experience Marlin does not like space in long names - M23 seems not handling them well

kleffa, 2024-05-06:
but M21;M23 04/CE3-SH~1.GCO 13470718 /CE3-Shogi_1xR-11h05m-66g.gcode 
works!
I try change name of one of folders, so it works now at least for this one
[image]

so yes Marlin has a bug and skip just numbers on Long file names, best reason to use shorts

closed 2024-05-06 by kleffa

luc-github, 2024-05-06:
@kleffa  Why not asking Marlin  to fix it ? I do not understand your approach

luc-github, 2024-05-06:
Additionnnaly on latest bug fix I can reporduce issue with missing sub folder containing numbers or letters
```
Begin file list
CUBE.GCO 112825 CUBE.GCO
INDEX_~1.GZ 91142 index.html.gz
LONGNA~1.GCO 18 longname.gcode
MY.GCO 144 MY.GCO
04/CUBE.GCO 112825 /CUBE.GCO
SUBDIR/CUBE.GCO 112825 /CUBE.GCO
End file list
ok
```

kleffa, 2024-05-07:
well I didn´t noted Marlin bug, I repoted after set this close, thanks for help!

just for asking, in latest bugfix "M21;M23 longname.gcode" works?

luc-github, 2024-05-07:
Seems yes:
``M20 L
08:38:37.150 > Begin file list
08:38:37.194 > CUBE.GCO 112825 CUBE.GCO
08:38:37.194 > INDEX_~1.GZ 91142 index.html.gz
08:38:37.195 > LONGNA~1.GCO 18 longname.gcode
08:38:37.197 > MY.GCO 144 MY.GCO
08:38:37.212 > 04/CUBE.GCO 112825 /CUBE.GCO
08:38:37.225 > SUBDIR/CUBE.GCO 112825 /CUBE.GCO
08:38:37.229 > End file list
08:38:37.229 > ok
M23 longname.gcode
08:38:53.001 > echo:Now fresh file: longname.gcode
08:38:53.003 > File opened: longname.gcode Size: 18
08:38:53.006 > File selected
08:38:53.009 > ok
M24
08:39:00.300 > ok
08:39:00.303 > Done printing file
```

luc-github, 2024-05-07:
but be noted sub dir is not working :
```
M23 04/CUBE.GCO
08:41:37.319 > echo:Now fresh file: 04/CUBE.GCO
08:41:37.320 > open failed, File: 04.
08:41:37.320 > open failed, File: 04/CUBE.GCO.
08:41:37.323 > ok
```

kleffa, 2024-05-07:
isn´t really to complex make a file explorer with M20 L that show Long name, but play short name? I noted is not just me with this kind of troubles

luc-github, 2024-05-07:
no but it will break all longname support implemented in Marlin for shared SD used by ESP3DLib, which means need to review all process for Marlin and long name in WebUI 
And I have no time for workaround an issue that should be fixed upstream