luc-github/ESP3D-WEBUI #392: [BUG] BTT TFT SD/USB listing fails because of M21

rondlh, 2024-06-19:
**Describe the bug**
Trying to list the files on the SD/USB port of an attached BTT TFT35 V3.0 fails.
An errors is reported: "Parse response error, format is incorrect".
Changing the TFT type in the Settings/Files panel does not solve this issue. (Removing the M21 solves the issue for me).
I believe that at least for the BTT Touchscreen firmware the M21 is not needed.

**To Reproduce**
Steps to reproduce the behavior:
1. Try to list the files of an attached BTT TFT SD/USB card in the dashboards
This fails with: 
```
M21
M20 SD:/
echo:No SD card
Media Init Fail
Begin file list
ALIGN_XY.GCODE
End file list
ok
```

OR:

```
M21
M20 U:/
echo:No SD card
Media Init Fail
Begin file list
IDEX_COIN.gcode
End file list
ok

```

As can be seen, an M21(init media) is send before the M20(list media). If I remove the M21 then things work for me.

**Expected behavior**
Works like before (like a few months ago).

**ESP3D Firmware:  (please complete the following information)**
- Version: 3.0.0-a234
- Web UI Version: 3.0.0-a64.M2.1]

**Target Firmware:  (please complete the following information)**
 - Name: Marlin
 - Version: Latest bugfix

**Board used (please complete the following information):**
 - MCU: ESP32]
 - Name: Development module
 - Flash size: 4M

**Browser used (please complete the following information):**
- Name: Chrome
- Version: 126
- Operating system: Windows 10

luc-github, 2024-06-19:
Yes put M21 in TFT commands may lead to failure if no SD in main board SD card reader - I guess I did the test with SD in reader and missed the issue - thank you 
I have updated the default commands in settings for BTT SD and USB :
[image]
I have rebuild the package but if the commands are already saved in preferences.json need to update them manually in settings/interface or delete preferences.json and reload webUI

luc-github, 2024-06-19:
I close issue as fixed - thank you for pointing out

closed 2024-06-19 by luc-github

luc-github, 2024-06-19:
About title I try to make it as simple as possible 
people who flash MKS TFT with BTT FW compatible should understand what they are doing, and they are not a majority of user (I did it too before) 

FYI I already got comment like:  what means: `Update FW`  and after  udate it what means `Update Firmware`  such thiing is never ending unfortunatly so Use `TFT Firmware` may confuse them ...

rondlh, 2024-06-19:
All clear, thanks again for the quick response and support!