luc-github/ESP3D-WEBUI #394: Add option for files to default to SD instead of Flash

jonathjon, 2024-06-30:
Please use this issue only for esp3d.io web site issue 
for [ESP3D](https://github.com/luc-github/ESP3D) /  [ESP3DLib](https://github.com/luc-github/ESP3DLib) /  [ESP3D-TFT](https://github.com/luc-github/ESP3D-TFT)  /  [ESP3D-WEBUI](https://github.com/luc-github/ESP3D-WEBUI)

Please describe what would be an enhancement

Flash is rarely used though WebUI 3, but SD card is used every time I run my CNC machine.  Currently the files section defaults to flash at startup.  Can you add an option to default to SD instead of flash?  

Thank you!

jonathjon, 2024-06-30:
Thank you.  I understand what you are saying and appreciate your time.  That is currently what I do, and then when it comes time to update I completely forget its there.  That's what gave me this idea.  I'm no programmer that's for sure so if its possible that's awesome, if its not then no big deal.  Never know unless you ask lol.  Thanks again!

luc-github, 2024-06-30:
to update flash you can use the update button
[image]

on some system like  grblHAL there is no other way to update webui because flash is not available

luc-github, 2024-07-02:
Fixed by https://github.com/luc-github/ESP3D-WEBUI/commit/423dbae646ada07fe3088f0e66b46ab671b21491

closed 2024-07-02 by luc-github