luc-github/ESP3D-WEBUI #397: [BUG] Odd Terminal history browsing behavior

rondlh, 2024-07-12:
**Describe the bug**
Using UP and DOWN to scroll through the previously send terminal commands/history doesn't work reliably.
1. It seems the history browsing is affected by new lines being added to the terminal screen.
I continuously have temperature reports coming in on the terminal, this seems to cause issues.
If I press UP then I get the previous command that was sent, but when I press UP again, and a new lines of data arrived in the terminal screen (before I pressed UP for the second time), then the same command is shown, not the command before the previous command.

2. Pressing DOWN is also affected in the same way as pressing up, and it seems that I need to press DOWN twice before I get the expected command.

**To Reproduce**
Steps to reproduce the behavior:
1. Send some command (M117 1, M117 2, M117 3)
3. Press UP (M117 3 should be shown)
4. Wait until a new lines arrives in the terminal screen
5. Press UP again, I would expect "M117 2", but I get "M117 3"

**ESP3D Firmware:**
- Version: 3.0.0.a234
- Web UI Version: 3.0.0-a68-M2.1
- Wifi mode: Client mode

**Target Firmware:**
 - Name: Lastest Marlin 2.1.x bugfix

**Board used:**
 - MCU: ESP32 Dev module
 - Name: NodeMCU V1.2
 - Flash size: 4M

**Browser used:**
- Name: Chrome 126.0.6478.127
- Operating system: Windows 10

luc-github, 2024-07-12:
Yes you are right, the variable of the monitor history index is reset at each refresh, purpose was to reset the history when terminal is no more displayed but I did it the wrong way, I have forget any refresh will have same effect orz...
that is now fixed (a69), I have also added another way to reset history index to latest: just enter empty line 

it is https://github.com/luc-github/ESP3D-WEBUI/commit/0334b1169568b53a6615aeff03788731fa56cb3e

rondlh, 2024-07-13:
> Yes you are right, the variable of the monitor history index is reset at each refresh, purpose was to reset the history when terminal is no more displayed but I did it the wrong way, I have forget any refresh will have same effect orz... that is now fixed (a69), I have also added another way to reset history index to latest: just enter empty line
> 
> it is [0334b11](https://github.com/luc-github/ESP3D-WEBUI/commit/0334b1169568b53a6615aeff03788731fa56cb3e)

Wow, so fast.
The UP key now works as expected, but I still have to press DOWN twice for it to work.

> I have also added another way to reset history index to latest: just enter empty line

That works well, excellent!

luc-github, 2024-07-13:
the Up / Down key was working for me so I had some time to reproduce the issue and fixed - it is fixed in a70

rondlh, 2024-07-13:
> the Up / Down key was working for me so I had some time to reproduce the issue and fixed - it is fixed in a70

Working, thanks, and great job again!

rondlh, 2024-07-13:
Fixed, thanks, and great job again!

closed 2024-07-13 by rondlh