luc-github/ESP3D-WEBUI #402: [Question] How to print uploaded file?

probonopd, 2024-12-18:
Now that I have uploaded a GCODE file, how do I actually send it to the printer over USB? 

[image]

Note that I can communicate well with the printer (e.g., move the extruder). But I am missing the "Print" button next to the GCODE file.

luc-github, 2024-12-18:
As mentioned in https://esp3d.io/ESP3D/Version_3.X/ `The firmware must be used with [ESP3D-WEBUI 3.X](https://esp3d.io/ESP3D-WebUI/Version_3.x/)`
so you need to go to https://esp3d.io/ESP3D-WebUI/Version_3.X/installation/

luc-github, 2024-12-18:
> But I am missing the "Print" button next to the GCODE file.

You cannot print from flash - streaming is not implemented in esp3d yet

closed 2024-12-18 by luc-github

probonopd, 2024-12-19:
Thanks @luc-github. Can we turn this into a feature request then, in case one doesn't already exist for it?

luc-github, 2024-12-19:
it is already https://github.com/luc-github/ESP3D/issues/243