luc-github/ESP3D-WEBUI #408: Unable to activate uploaded CSS

stefan63, 2025-02-22:
**Describe the bug**
I'm runing FluidNC and WebUI3 (Latest). 
I have created a CSS file that I have tried to upload as test, test1. test2 etc but they dont shoe up in the.
Esp3D Interface page in webUI3. I can se the fiules are there in dashboard/files/Flash.

Removing all the theme files and uploading my own test file again didn't solve the issue.
After Removing all files again  and uploading my test file now renamed to dark (like one of the example files) now it suddenly are working. Dod refresh pages after all upload cycles.

luc-github, 2025-02-22:
name of theme have a special format to be displayed
did you read https://esp3d.io/ESP3D-WebUI/Version_3.X/documentation/themes/ ?

stefan63, 2025-02-22:
> name of theme have a special format to be displayed did you read https://esp3d.io/ESP3D-WebUI/Version_3.X/documentation/themes/ ?

Thank you I was looking for it but didnt find it :-)
Will share my CSS if it turns out nice (If yo want me to ;)

Kind regards/Stefan

luc-github, 2025-02-22:
ok, have fun, I close issue then

closed 2025-02-22 by luc-github