luc-github/ESP3D-WEBUI #411: [FEATURE REQUEST] Flexible panels customizations to optimize empty space

sstefanov, 2025-03-30:
**Is your feature request related to a problem? Please describe.**
Current screen layout contains a lot of unused space. When more panels are visible the screen must be scrolled up/down to see all of them.
For example, Print status panel contains only 2 lines and all other space is empty.

**Describe the solution you'd like**
I have 2 proposals:
1. Option to resize panels to fit to contained items.
2. Option to move items to other panels. For example, all items from Temperatures panel can fit well in empty space in Printer status panel or vice-versa.

This way screen space can be used a lot more effective.

luc-github, 2025-03-30:
PR is welcome for such solution because  I do have time to spend on this

sstefanov, 2026-09-25:
Hi,
I saw panel size, but it is only about the width.
Will be excellent if minimum height also can be added.

luc-github, 2026-09-25:
it is maximum height not minimum, and it is just above  width,  you cannot miss it

[image]